open-rdc/Creality-K1-Slicer-Profiles #1: orcaslicerの設定ファイルを作成する

MibuchiYuta, 2024-03-12:


MibuchiYuta, 2024-03-13:
orcaslicerの設定ファイルはデフォルトで用意されているものをベースに変更していく。

MibuchiYuta, 2024-03-18:
フィラメントは kexcelled のplaを使用する

MibuchiYuta, 2024-03-18:
@hellocit ベンチマークモデルの候補を早めに決定してください

hellocit, 2024-03-18:
@MibuchiYuta 以前，紹介していただいたベンチマークモデルのgithubページを以下に記載しました．3D Benchyと比較して，見るべき指標が分かりやすいと考えるのでkickstarter-autodesk-3dでよいかなと考えますがいかかでしょうか．

https://github.com/kickstarter/kickstarter-autodesk-3d

hellocit, 2024-03-18:
**方針**
デフォルトの設定を用いて印刷時間がどれほどかかるか，次に，粗め，細かめにした時の3D プリンタの挙動や造形物について観察する．

MibuchiYuta, 2024-03-18:
> @MibuchiYuta 以前，紹介していただいたベンチマークモデルのgithubページを以下に記載しました．3D Benchyと比較して，見るべき指標が分かりやすいと考えるのでkickstarter-autodesk-3dでよいかなと考えますがいかかでしょうか．
> 
> https://github.com/kickstarter/kickstarter-autodesk-3d

大丈夫です

MibuchiYuta, 2024-03-18:
> **方針** デフォルトの設定を用いて印刷時間がどれほどかかるか，次に，粗め，細かめにした時の3D プリンタの挙動や造形物について観察する．

@hellocit こちらお忘れなく

> 成功時の空間の温度と湿度と天気をメモする

hellocit, 2024-03-18:
承知しました．

hellocit, 2024-03-18:
https://docs.google.com/document/d/12Ni78JbNmHuk7BiG3rXoD1zSS72fnm0MQ8EULhiQKTo/edit?usp=sharing

メモのひな型を上記に示す．気になることがあれば記入する．

MibuchiYuta, 2024-03-31:
@hellocit 情報を分散させないでください。issueに気軽に書き込んで大丈夫です。

MibuchiYuta, 2024-03-31:
@hellocit #12 の問題が解決しましたので、作業を開始してください。最初の作業確認の期限ですが、4/5まででお願いします。

MibuchiYuta, 2024-03-31:
設定ファイルの詰めかたについて検討しました。参考にしてください。

## 1 マシンスペックの確認
流量と印刷速度の最大を計測し、それを基準にする。
orcaslicerのキャリブレーションメニューから確認できます。(Max Volmetric speedとVFA。VFAはwiki情報なし [詳細](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/12#issuecomment-2028827763))
## 2 高速印刷パラメータの作成
安定して動作する範囲内で高速印刷用のパラメータを作成する。0.3mmピッチで移動速度は600mm/s 20000mm/s^2、印刷速度などは**1**を参考に。orcaslicerのキャリブレーションメニューには各種パラメータのチューニング用データがあるのでそれも利用してください。
## 3 綺麗な印刷パラメータ
高速印刷用のパラメータをもとに速度を落とし、時間よりも品質を重視して速度のデチューンを行う。速度の他にも印刷品質に影響するパラメータはあるためそちらも検証する。このパラメータはどうしても綺麗に印刷したい時のみ利用するイメージです。
## 4綺麗な印刷パラメータ用のサポート追加
**3**のパラメータを元にサポート用のパラメータにしていく。サポート用のベンチモデルも[printables](https://www.printables.com/model/260191-support-bench-support-test)などにあるのでそれを利用する。複数のモデルでチェックすることを推奨します。
## 5高速印刷パラメータ用のサポート追加
**4**と同じ

hellocit, 2024-04-03:
# 印刷結果 240403_1949

# 印刷するため追記を行う

湿度：53%
温度：20.3
天気：雨＊１

## 印刷目的
高速印刷パラメータの作成を行うために積層ピッチ，移動速度と加速度に変更を加えて，印刷時間がどれほど短くなるかを記録する．

## 印刷条件
変更したパラメータに関して，以下のjsonファイルに示す．
フィラメント： kexcelled 黒

## 印刷結果
印刷時間：1h18m
後ほど追記を行う.

## 備考
・印刷開始時の湿度と温度を記録した．
＊１天気は[このサイト](https://weather.yahoo.co.jp/weather/12/4510/12204.html)を参考にした．

hellocit, 2024-04-03:
# 今後の方針
- vfaなどから破綻しない程度に最速を目指す．
1. マシンスペック最大を目指す．
2. 積層高さは0.3[mm]とする．

加えて，今後の3Dプリンタを扱う人のために過程や議論を可能な限り全て情報を記載する．

hellocit, 2024-04-04:
@MibuchiYuta VFAのモデルはどのように出力すればよいでしょうか．基本的には以下に添付するキャリブレーションのボタンとその種類により，出力されるモデルが変更されると思います．VFAに関して，どのように出力すればよいか教えていただけると助かります．
[image]

hellocit, 2024-04-04:
[この件](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2036179743)に関して，VFAについてはマシンスペックの最大値でよいことが確認できているため，そのようにする．

hellocit, 2024-04-04:
# 印刷結果 240404_1336

湿度：56%
温度：18.9
天気：曇り＊１

## 印刷目的
[高速印刷パラメータ](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2028835841)の作成を行うために壁面，速度，加速度に変更を加えて，印刷時間がどれほど短くなるかを記録する．

## 印刷条件
変更したパラメータに関して，以下のjsonファイルに示す．
[default_param_first_fast_ver2.json](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/files/14861808/default_param_first_fast_ver2.json)
フィラメント： kexcelled 黒

## 印刷結果
印刷時間：1h7m
[image]
[image]

## 感想
[この試行](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2033982843)と比較すると印刷のきれいさは落ちていると感じる．

## 備考
・印刷開始時の湿度と温度を記録した．
＊１天気は[このサイト](https://weather.yahoo.co.jp/weather/12/4510/12204.html)を参考にした．

hellocit, 2024-04-04:
**今後の方針**
モデルを変更し，3D Benchyを使用する．3D benchyを使用して，パラメータが煮詰まったらkickstarter-autodesk-3dを使用する．

kyo0221, 2024-04-04:
# 印刷結果_240404_2112

湿度 : 60%
温度 : 15.6℃
天気 : くもり

## 印刷目的
[高速印刷パラメータ](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2028835841)の作成を行うために壁面，速度，加速度に変更を加えて，印刷時間がどれほど短くなるかを記録する．

## 印刷条件
フィラメント： kexcelled 黒

## 印刷結果
印刷時間：15.14m
[image]
[image]

## 感想
原型は保っているが1層目に隙間ができているように感じた.

## 備考
・印刷開始時の湿度と温度を記録した．
＊１天気は[このサイト](https://weather.yahoo.co.jp/weather/12/4510/12204.html)を参考にした．

hellocit, 2024-04-10:
[こちら](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2028835841)の設定ファイルの詰め方について1 マシンスペックの確認から行う．理由は，私自身の理解を深めることとK1 maxのハードウェアの限界を調査するためである．

hellocit, 2024-04-10:
VFAについて以下のパラメータで印刷を行う．
Start speed:50mm/s
End speed:650m/s
step:25m/s

[image]

hellocit, 2024-04-14:
Max flowrateについて以下のパラメータで印刷を行う．
Start volumetric speed:5 mm^3/s
End volumetric speed:20 mm^3/s
step:0.5 mm^3/s

[image]

hellocit, 2024-04-14:
VFAのテストについて記述する．
写真

hellocit, 2024-04-14:
# 追記を行う

Max flowrateのテストについて記述する．[このコメント](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/issues/1#issuecomment-2053925688)このコメントで記載したパラメータで印刷を行った．

# 印刷結果 240414

湿度：54%
温度：21.2

## 印刷目的
既定のパラメータで印刷したときにどのような印刷結果になるかを記録する．

## 印刷条件
[使用したプロジェクト](https://drive.google.com/file/d/1HiwHJfZXXHXEyAQggb0tv-twmWi361s9/view?usp=sharing)は上記のリンクにあるがOrcaSlicerの所定のパラメータで印刷を行っているため，意図的に変更したものは以下のパラメータである．
```
Start volumetric speed:5 mm^3/s
End volumetric speed:20 mm^3/s
step:0.5 mm^3/s
```
フィラメント： kexcelled 黒

## 印刷結果
32mmすべてにわたって正常に印刷OKであることを確認，以下の計算式を用いて最大体積速度を計算するがheightをどこからもってくればよいかわからない．
start + (height-measured * step)
5+(height-measured*0.5)

## 考察

MibuchiYuta, 2024-06-09:
暫定的な印刷パラメータをcomitしました。以下のやり方で作業しました。

1962ce5d37771ab25ebbd5ee8c85221b100aaec2

> 設定ファイルの詰めかたについて検討しました。参考にしてください。
> 
> ## 1 マシンスペックの確認
> 流量と印刷速度の最大を計測し、それを基準にする。 orcaslicerのキャリブレーションメニューから確認できます。(Max Volmetric speedとVFA。VFAはwiki情報なし [詳細](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#Advanced-Calibration))
> 
> ## 2 高速印刷パラメータの作成
> 安定して動作する範囲内で高速印刷用のパラメータを作成する。0.3mmピッチで移動速度は600mm/s 20000mm/s^2、印刷速度などは**1**を参考に。orcaslicerのキャリブレーションメニューには各種パラメータのチューニング用データがあるのでそれも利用してください。
> 
> ## 3 綺麗な印刷パラメータの作成
> 高速印刷用のパラメータをもとに速度を落とし、時間よりも品質を重視して速度のデチューンを行う。速度の他にも印刷品質に影響するパラメータはあるためそちらも検証する。このパラメータはどうしても綺麗に印刷したい時のみ利用するイメージです。
> 
> ## 4綺麗な印刷パラメータ用のサポート追加
> **3**のパラメータを元にサポート用のパラメータにしていく。サポート用のベンチモデルも[printables](https://www.printables.com/model/260191-support-bench-support-test)などにあるのでそれを利用する。複数のモデルでチェックすることを推奨します。
> 
> ## 5高速印刷パラメータ用のサポート追加
> **4**と同じ

MibuchiYuta, 2024-06-09:
また、このコミットでは壁の厚みを増やしたパラメータも追加してあります。

closed 2024-06-09 by MibuchiYuta