open-rdc/Creality-K1-Slicer-Profiles #2: 環境構築手順をwikiにまとめる

MibuchiYuta, 2024-03-12:


MibuchiYuta, 2024-03-31:
@Ryusei-Baba 
WindowsとUbuntuでのインストール方法に関してwikiの作成をお願いします。4/5までにできますか？

MibuchiYuta, 2024-03-31:
@MibuchiYuta 
Inventorでのstlファイル出力とプリンターの準備・開始・後片付けのwiki作成、homeの整備を行います。

MibuchiYuta, 2024-04-02:
## markdown記法について

[チートシート](https://github.com/MibuchiYuta/git_memo/blob/master/github/markdown.md)を作成したので参考にどうぞ

hellocit, 2024-05-14:
@MibuchiYuta  質問です．下記のwikiページについて何を書けばよいか議論をしたいです．質問の意図は他のwikiやissueと役割がかぶるのを防ぎたいです．

[wikiページ](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/wiki/%E3%83%97%E3%83%AA%E3%83%B3%E3%82%BF%E3%83%BC%E3%81%AE%E9%96%8B%E5%A7%8B,-%E5%BE%8C%E7%89%87%E4%BB%98%E3%81%91%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6)

MibuchiYuta, 2024-05-14:
ただgcodeを送るのではなく、ベッドの用意や動かす前の確認事項、印刷後の取り外し方や後片付けを記載してください

hellocit, 2024-05-14:
承知しました．ありがとうございます．それらに関して記載します．

MibuchiYuta, 2024-06-09:
設定ファイルをコミットしたのでこのリポジトリのクローンとファイルの読み込み方法を記載して欲しいです。

hellocit, 2024-06-12:
@MibuchiYuta 
こちらのissueに関して[プリンタの開始, 後片付けについて](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/wiki/%E3%83%97%E3%83%AA%E3%83%B3%E3%82%BF%E3%81%AE%E9%96%8B%E5%A7%8B,-%E5%BE%8C%E7%89%87%E4%BB%98%E3%81%91%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6)
と
[リポジトリのクローンと設定ファイルのロード](https://github.com/open-rdc/Creality-K1-Slicer-Profiles/wiki/%E3%83%AA%E3%83%9D%E3%82%B8%E3%83%88%E3%83%AA%E3%81%AE%E3%82%AF%E3%83%AD%E3%83%BC%E3%83%B3%E3%81%A8%E8%A8%AD%E5%AE%9A%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB%E3%81%AE%E3%83%AD%E3%83%BC%E3%83%89)が終了していると思うので承認さえしていただければissueを閉じたいと思うのでよろしくお願いします。

MibuchiYuta, 2024-06-13:
Homeページの編集を私がしたらcloseします

MibuchiYuta, 2024-06-14:
書きましたのでcloseします
https://github.com/open-rdc/Creality-K1-Slicer-Profiles/wiki

closed 2024-06-14 by MibuchiYuta