open-rdc/Creality-K1-Slicer-Profiles #19: エクストルーダ移動時の異音

hellocit, 2024-05-24:
表題のとおり異音がする．

hellocit, 2024-05-24:
https://github.com/open-rdc/Creality-K1-Slicer-Profiles/assets/91714744/954c4e2e-e529-4564-8e2e-1a73e89ba1b6

hellocit, 2024-05-24:
印刷は順調であるため，問題でない可能性もある．

hellocit, 2024-05-24:
[image]
[image]

hellocit, 2024-05-28:
@MibuchiYuta @kyo0221 こちらのissueに関して問題なければ閉じたいのですがよろしいですか？

MibuchiYuta, 2024-05-28:
音が鳴る理由の追求だけお願いします。その内容で後回しにするか判断しましょう。

hellocit, 2024-05-30:
@MibuchiYuta 調査が遅れました申し訳ないです．

## 調査結果
添付の動画より，xとy軸方向に対してエクストルーダが移動しているときに異音がすることが判明した．具体的にどの部品から異音がするかについては添付した写真の赤丸の部品であると結論付ける．異音がなる理由に関しては，プリンタ自体の分解をしたわけではないので推測になるが赤丸の部品の組付けがよくないからという理由が考えられる．

https://github.com/open-rdc/Creality-K1-Slicer-Profiles/assets/91714744/c3ff15d0-9864-4a22-9ad2-9bffa449a99d

[image]

MibuchiYuta, 2024-06-16:
稼働しているものは無給油ブッシュが使われているようです。ロッドも含めてメンテナンス方法があるはずです。

MibuchiYuta, 2024-06-16:
メンテナンス方法がわかれば
 #3 で議論したいですね。

あとこちらのissueの題名ですがエクストルーダーとしてしまうとフィラメントの押し出し機構に問題があるように思えてしまうので、ヘッドの移動音とかCoreXYの異音とかにした方がいいと思います．

hellocit, 2024-06-16:
ありがとうございます。issueの名前を変更しました。

- [こちらのissueと同じと思われる問題1](https://www.reddit.com/r/Creality/comments/1512tdp/my_creality_k1_printer_is_making_a_clicking_noise/)
- [こちらのissueと同じと思われる問題2](https://www.reddit.com/r/crealityk1/comments/18akryb/clicking_on_belts_i_think/)
ある程度調査しましたが同じ問題にぶつかっている人はいました。文章を用いた対策は示されていますが動画を用いたメンテナンス方法を提示してくれている人はいませんでした。

MibuchiYuta, 2024-06-20:
今後どうしますか?

hellocit, 2024-06-20:
1ヶ月こちらの問題に対する対応が取られていないにも関わらず致命的な問題は引き起こされていないので本issueはクローズしても問題ないと考えています。しかし、ベルト等のメンテナンスに繋がる内容ではあると思いますのでそちらで言及したいと考えています。

@MibuchiYuta 上記の対応でいかがでしょうか。

MibuchiYuta, 2024-06-20:
わかりました

closed 2024-06-20 by MibuchiYuta