open-rdc/Creality-K1-Slicer-Profiles #29: ノズルが加熱できない問題

kyo0221, 2024-12-12:
フィラメントが詰まってしまう問題が発生し, ノズル部分にフィラメントが詰まっていると考えて加熱しようとしたが, エラーが表示されて加熱できない問題が発生した.
[image]

kyo0221, 2024-12-12:
ノズルの配線がちぎれていることを確認した.

[image]

kyo0221, 2024-12-13:
修理ができないと判断して, 再購入することにしました.

購入品
https://amzn.asia/d/6HCT4zc

kyo0221, 2024-12-13:
問題の経緯

・3Dプリンターのパラメータチューニングのため, 最大流量の調整をするベンチマークを印刷している途中でフィラメントが出てこなくなっていた

・フィラメントを外し, エクストルーダー部分に汚れ等が溜まっていないか確認した

・汚れ等は認められなかったため, ノズル部分で詰まっていることを疑い, フィラメントがノズル部分に定着していたため加熱しようとした.

・上記のノズルが加熱できない問題が発生した

ryotaema, 2024-12-26:
購入したノズルに付け替えたところ加熱できるようになりました。
詰まりも加熱したところ解消されていました。
パラメータチューニングに移行します。

closed 2024-12-26 by ryotaema